Zhikharev/selen #2: RISC5 ISA version

Zhikharev, 2015-10-12:
На какую версию isa мы ориентируемся: v2.0?

gipnocow, 2015-10-12:
Да, на версию, которая была загружена с аккаунта Болотникова

Zhikharev, 2015-10-12:
У нас там 36 инструкций, а в официальном документе 47. Как можно это прокомментировать? 
Если на данном этапе мы решили, какие-либо инструкции не реализовывать, то их нужно отметить.

AlexBolotnikov, 2015-10-12:
Да на 2 версию (там есть все кроме команд для операционной системы)
--
С уважением Александр Болотников Monday,

Zhikharev, 2015-10-13:
Нужно удалить из репозитория все лишнее файлы с isa, оставить одну конечную

Zhikharev, 2015-10-13:
Кроме того возник ряд вопрос по действиям в документах, их нужно в тексте подправить:
1) ADD. Что там с переполнением?
2) SLTU. Что за rd2?
3) ADDI. Что такое se? Как рассчитывается?
4) Какие операции совершаются при ветвление (что к чему прибавляется)?
5) В ISA нужно указать имена регистров и переводы сокращений

Jettatura, 2015-10-14:
Придется отлаживать, работать с этим в чем-нибудь автоматическом 
поэтому:
на первом этапе надо обязательно включать
NOP. ( убирать прыжки, отлаживать) 
что-нибудь вроде  SBREAK, если вдруг мы захотим  знать что у нас выполнение доходит до интересных точек, или до конца.
Без этого сложность написания чего-то что эти инструкции кушает повышается.

Jettatura, 2015-10-14:
просто я их в ISA_version_2 не увидел, а полного RISC V нам сейчас наверное и не надо.
Отметьте их, пожалуйста.

Zhikharev, 2015-10-14:
1) По поводу NOP:
The NOP instruction does not change any user-visible state, except for advancing the pc. NOP is encoded as ADDI x0, x0, 0.
2) SBREAK будем думать

closed 2015-11-04 by Zhikharev