Zhikharev/selen #3: Коммутатор

Zhikharev, 2015-10-13:
Информация к размышлению. В целом коммутатор не связан с нашими блоками периферии (iohub, ram, rom) и с ядром. Он просто имеет определённое количество портов. Приоритетность портов и коммутируемость может быть настроена через программной доступные регистры коммутатора встроенным софтом. 
Я не призываю делать универсальный коммутатор сразу, сначала нужно получить готовое решение для наших нужд. Но иметь в виду стоит.

pointhex, 2015-10-13:
Я так понял, коммутатор должен быть полноценным устройством со своими ригистрами и с несколькими портами для слейвов и несколько для мастеров?
3 слева надо и 3 мастера?

Zhikharev, 2015-10-13:
Вообще, у полноценного коммутатора все порты должны быть равнозначны, но тут мы делаем конкретную реализацию.
Вопрос терминологии. Если к порту подключено мастер устройство, то порт будем называть мастер-портом. Получается коммутатор может коммутировать данные с мастер порта на слейв порт. 

Предлагаю сигналы коммутатора не привязывать к именам блоков СнК, но создать дополнительную обёртку (дополнительный модуль) с сигналами соответствующими cpu, ram и т.д., внутри которой установить коммутатор и правильно соединить.

pointhex, 2015-10-13:
Хорошо сделаю обертку. 
А есть смысл делать нормальный коммутатор с 3 мастерами и слейвами? Или пока оставить как есть?

Zhikharev, 2015-10-13:
Если делать более строго, с определённым количеством слейвов и мастеров, то нужно добавлять программное управление арбитражом и т. п.

pointhex, 2015-10-13:
Надо будет лично все по точнее обсудить.

Zhikharev, 2015-10-13:
К субботе готовьте вопросы и варианты решения, в институте обсудим.

pointhex, 2015-10-13:
Хорошо. Во сколько?

closed 2015-11-04 by Zhikharev