Zhikharev/selen #7: Правки

Zhikharev, 2015-12-07:
В rtl описании:
1) названия всех параметров дать капс-локом
2) не использовать магические числа, например кодировки заголовка: ввести локальные параметры

pointhex, 2015-12-08:
Хорошо исправлю.

pointhex, 2015-12-09:
Я поправил оформление.

Zhikharev, 2015-12-09:
Ошибки при компиляции тестов для системы

Back to file '/media/sf_selen.git//verif/system/system_tb_top.sv'.
Parsing included file '/media/sf_selen.git//rtl/io_hub/rtl/io_hub.v'.

Error-[SE] Syntax error
  Following verilog source has syntax error :
  "/media/sf_selen.git//rtl/io_hub/rtl/io_hub.v", 28: token is ','
                output                           ,

Zhikharev, 2015-12-09:
Error-[SE] Syntax error
  Following verilog source has syntax error :
  "/media/sf_selen.git//rtl/io_hub/rtl/io_hub.v", 90: token is 'begin'
                        if ((~empty) & (status_reg[1]) begin  //
  state_finish

Zhikharev, 2015-12-09:
Error-[UPIMI-E] Undefined port in module instantiation
/media/sf_selen.git//rtl/io_hub/rtl/io_top.v, 65      
  Port "dma_cyc_o" is not defined in module 'io_hub' defined in 
  "/media/sf_selen.git//rtl/io_hub/rtl/io_hub.v", 12            
  Module instance: io_hub io_hub( .clk (sys_clk),  .rst (sys_rst),  .data 
  (data),  .status_reg (status_reg),  .addr_first_reg (addr_first_reg),   
  .addr_end_reg (addr_end_reg),  . ...                                    

Error-[UPIMI-E] Undefined port in module instantiation
/media/sf_selen.git//rtl/selen_top.sv, 91             
  Port "dma_cyc_o" is not defined in module 'wb_comm' defined in 
  "/media/sf_selen.git//rtl/memory_commutator/rtl/wb_comm.v", 12 
  Module instance: wb_comm comm( .sys_clk (sys_clk),  .sys_rst (sys_rst),  
  .cpu_inst_stb_i (cpu_wbi_stb),  .cpu_inst_ack_o (cpu_wbi_ack),           
  .cpu_inst_cyc_i (cpu_wbi_cyc),  .cpu_i ...

Zhikharev, 2015-12-09:
Посыл такой: запускать мэйк (или адаптированный мэйк под modelsim) questaMake.
В общем случае это может выглядеть так:
make -f /media/sf_selen.git/verif/system/questaMakefile GIT_HOME=/media/sf_selen.git/ sim

Zhikharev, 2015-12-11:
Что с этим? Мы не может двигаться дальше пока система не собирается.

pointhex, 2015-12-11:
Я исправил.
Сам пока модел сим не запускал.
Я бы мог сегодня все поставить, а завтра часов 10:30-11:00 могли бы мне показать как там все запускать?

Zhikharev, 2015-12-11:
Ты же хотел убрать dma_cyc_o или он всё такие используется?

pointhex, 2015-12-11:
Он есть, но висит в воздухе, я пока оставлю.

closed 2015-12-11 by Zhikharev