Zhikharev/selen #8: Исправления в ISA

Zhikharev, 2015-12-09:
1) AUIPC: 0100011 -> 0010111
2) LUI: 0100011 -> 0110111
3) SB-type: 0010011 -> 1100011
4) JAL: 00010111 -> 1101111

Zhikharev, 2015-12-09:
Нужно проверить остальные коды в нашем документе, если обнаружишь ошибки, то отметить тут и исправить в документе.

Zhikharev, 2015-12-09:
Там вообще почти все не сходятся! Откуда были взяты эти коды?
Я смотрю значения в http://riscv.org/download.html#tab_spec_user_isa стр. 50

AlexBolotnikov, 2015-12-10:
В том документе что вы прислали на странице 50 ни чего нет про opcodes и все оп коды которые в нашей isa мы взяли из головы потому что в документе нет (ну или мы не нашли) кодировки оп кодов. насчет imm я поправлял может просто не залил(я еще раз посмотрю и залью исправденный вариант если это нужно)

Zhikharev, 2015-12-11:
Так что в итоге?

P.S. По каждой issue должно быть заключение

Zhikharev, 2016-01-11:
неактуально

closed 2016-01-11 by Zhikharev