Zhikharev/selen #11: Чтение из ROM

Zhikharev, 2015-12-11:
Сейчас в системной сборке процессор начинает отправлять запросы, но они не доходят до ROM и не выставляется сигнал stall -> процессор отправляет запросы до бесконечности.

Нужно исправить коммутатор и блоки периферии.

pointhex, 2015-12-11:
Я исправил.
Проверил в своем тесте.

Zhikharev, 2015-12-12:
Проблема не устранена. В ядро не возвращаются ответы и не выставляется stall

pointhex, 2015-12-12:
Значит буфер не успевает заполняться. Или он обязательно должен
заполняться? Тогда нужно вводить ещё один внешний сигнал РОМ для
управлением rd_en для fifo.

Zhikharev, 2015-12-12:
Буфер не обязательно должен заполняться, но сомнительно что проблема в этом.
Сколько позиций в буфере?
И это уж точно никак не влияет на то, что ответы не приходят.

pointhex, 2015-12-12:
А я понял, ответы не возвращаются все же.
Хм, я посмотрю еще раз, но на моем тесте вчера все работало.

Zhikharev, 2015-12-12:
Если в твоём тесте была не вся сборка, то тут может быть, что угодно.

pointhex, 2015-12-12:
Я проверю еще раз и отпишусь.

Zhikharev, 2015-12-13:
Такими темпами, мы не успеем. Есть ли продвижения тут?

pointhex, 2015-12-13:
Да продвижения есть.
Я проверил у себя у меня все окей.
Я поставил questasim единственное у меня ругается на то, что не включен режим проверки лицензии. Не подскажете в чем проблема?

Zhikharev, 2015-12-13:
Какая именно ошибка? Скопируй сюда.
Возможно ты не создал лицензию для своего компьютера.

pointhex, 2015-12-13:
Я создал, когда запускаю гуй он мне пишет Unable to checkout a viewer license necessary for use of the modelsim graphical user interface. Vsim is closing.

Zhikharev, 2015-12-13:
А без графического режима запускается? (нужно убрать команду -c)

Zhikharev, 2015-12-13:
То есть наоборот нужно добавить команду -c в SIM_OPTS например

pointhex, 2015-12-13:
Ну я запускал просто vsim. И все он сразу в гуй запускает.
В каком файле -с добавить? или в переменных окружения?

Zhikharev, 2015-12-13:
vsim -novopt -do 'run -all' -c system_tb_top

Zhikharev, 2015-12-13:
Или воспользоваться наших мэйкфалом:
make -f <путь>/questaMakefile GIT_HOME=<путь к репозиторию> sim SIM_OPTS="-c"

pointhex, 2015-12-13:
Спасибо!
Я попробую чуть позже и напишу.

pointhex, 2015-12-15:
Попробовал то, что мне сказали, но безуспешно.

pointhex, 2015-12-15:
Попробовал поставить все заново но все так же.

Zhikharev, 2015-12-15:
Что значит безуспешно? 
Скорее всего ты неправильно генерируешь лицензию. 
Указываешь LM_LICENSE_FILE? Следуешь ли инструкции, которая была сделана для установки questasim?

pointhex, 2015-12-15:
Да все сделал как надо.
LM_LICENSE_FILE добавил.

Zhikharev, 2015-12-15:
Ну если всё сделал как надо, но то должно работать. 
Если не работает, то опять же: какая ошибка?

closed 2016-01-27 by Zhikharev