Zhikharev/selen #12: Ошибки в CPU

Zhikharev, 2015-12-12:
Запускать тест +TESTNAME=direct_test
По тесту сначала идёт загрузка данных в регистры из памяти (cpu_load_seq) затем исполнение простых инструкций (cpu_direct_seq).

1) На интерфейсе WBD хаос
2) На интерфейсе WBI, неправильная отработка сигнала stall

AlexBolotnikov, 2015-12-13:
Я считаю что все поправил

Zhikharev, 2015-12-13:
1) Протокол на WBI работает неверно: вначале идут несколько стробов stb с одним адресом
[100.0ns][WBI MON] CPU request addr=0
[120.0ns][WBI MON] CPU request addr=0
[140.0ns][WBI MON] CPU request addr=0
[160.0ns][WBI MON] CPU request addr=0
[180.0ns][WBI MON] CPU request addr=0

Ядро запрашивает несколько инструкций по одному адресу?

Если ядро работает не конвеййерно, то перед выставлением очередного stb оно должно дождаться ack (читай стандарт wishbone)

Почему падает cyc когда приходит stall? сycдолжен опускать только когда ядро не ждёт овтетов.

По тесту ядру возвращаются команды  LW, например:
[280.0ns][WBI MON] CPU ack LW rs1= 9 rs2= 8 rd= 5 imm=00000a48

но на интерфейсе WBD тишина.

Zhikharev, 2016-01-11:
неактуально

closed 2016-01-11 by Zhikharev