Zhikharev/selen #14: BIg endian, загрузчик

Jettatura, 2015-12-13:
Симулятор работает с тем что есть, однако вопросы.
1) транслятор gipnocow кладет инструкции в big edian. Так надо? 
2) загрузчик: что представляет собой исполняемый файл в нашем случае.  Дайте объективной конкретики.

Zhikharev, 2015-12-13:
Поясни пожалуйста более подробно свои вопросы, желательно с примерами.

Jettatura, 2015-12-13:
я беру файл gg.txt, запускаю скрипт main.py с тем конфигом что предлагается. Вот, получаю два файла out.bin и outf.bin. В out.bin находятся оттранслированные инструкции, a в outf.bin тоже самое, только  еще некие "хедеры".

Чтобы выполнить эти инструкции в симуляторе я должен загрузить их в память симулятора. 
Память это прото набор байт, адресное пространство.  Я должен знать где в каком месте памяти мне положить инструкции, а где данные. Короче, мне надо сделать образ из outf.bin. Эти хедеры по видимому нужны для указания где какую секцию (код, данные) класть.
Вот, названия этих хедеров, в том виде что есть сейчас, не очень понятны: data_header это наверное адрес по которому надо размещать начало секции данных, а что такое start_address, end_address , не понятно.

Короче, надо сделать загрузчик который превращает ваши бинарники в образ памяти. Чтобы его сделать надо знать что куда класть. 

Big endian:
Это про гнал на порядок байт в инструкции записанной в файл. Это снимается, я думаю это я ошибся)) Там все как надо.

Zhikharev, 2015-12-14:
Сейчас ассемблер как раз создаёт образ памяти: у нас прямо за инструкциями лежат данные. Програмист должен сам гарантировать, что процессор не начнёт исполнять неправильный код (бесконечный цикл в конце например)

Jettatura, 2015-12-14:
Я немного не про исполнение неправильного кода , это то да.

Вот пример:

Есть бинарник out.bin- оттранслированый файл gg.txt
вот его дизассемблер:
```
         0	  0x500813	ADDI	S0, ZERO, 0x5
       0x4	  0xc81813	SLLI	        S0, S0, 0xc
       0x8	  0x500413	ADDI	T0, ZERO, 0x5
       0xc	  0x300493	ADDI	T1, ZERO, 0x3
      0x10	  0x600513	ADDI	T2, ZERO, 0x6
      0x14	  0x9405b3	ADD	T3, T0, T1
      0x18	  0xb50633	ADD	T4, T2, T3
      0x1c	   0x82403	LW	T0, [S0 + 0]
      0x20	  0x482483	LW	T1, [S0 + 0x4]
      0x24	  0x882503	LW	T2, [S0 + 0x8]
      0x28	  0xc82583	LW	T3, [S0 + 0xc]
      0x2c	 0x1082603	LW	T4, [S0 + 0x10]
      0x30	 0x1182023	SW	S0, [S1 + 0]
      0x34	 0x1282223	SW	S0, [S2 + 0x4]
      0x38	 0x12888b3	ADD	S1, S1, S2
      0x3c	 0x1282423	SW	S0, [S2 + 0x8]
      0x40	 0x12888b3	ADD	S1, S1, S2
      0x44	 0x1282623	SW	S0, [S2 + 0xc]
      0x48	 0x12888b3	ADD	S1, S1, S2
      0x4c	 0x1282823	SW	S0, [S2 + 0x10]
      0x50	 0x12888b3	ADD	S1, S1, S2
```
Симулятор доходит до 8 инструкции, где надо загрузить в T0 слово из 0x5000.
Но размер бинарника всего 0x54.  Значит бинарник это не образ памяти.  Это просто инструкции.  Просто под образом я понимаю всю память, и которой обращаются тоже.

Давайте передавать симулятору опцию в командной строке устанавливающую  размер адресного пространства. Он будет класть бинарник в начало, а все остальное место считать допустимым.

Но тогда я не понимаю смысла файла с хедерами, outf.bin. Я думал что эти хедеры указывают где чего положить, и это и есть тот самый исполняемый файл.  
потому как файл с хедерами это те же инструкции  с дополнительными записями.

Zhikharev, 2015-12-14:
Сначала про хедеры:
1) Это просто алгоритмика для передачи программы от хоста на плату. IOhub получает по uart данные, анализирует хедеры и решает что делать с данными: отправлять в память или обновлять свои регистры.

Zhikharev, 2015-12-14:
По поводу пространства:
- в примере видимо ошибка, там указан коментарий "загрузим массив память", а используются команды LW
-при работе с асемблером можно создавать свои переменные с директивой data (раньше точно было), ассемблер поместит эти данные после инструкций

Пример:

LW $t0, p1
.data
p1 0x0000ffff

В итоге получится что-то вроде 

0 LW $t0, 0(4)
4  0x0000ffff

Да, будем передавать параметр для симулятора.

Ещё есть какие-то вопросы, которые нужно прояснить?

Jettatura, 2015-12-14:
круто, это оказывается ваши желязячные штучки. 

Я теперь все понял. 

Короче, симулятор пока будет работать так:

Запуск:
симулятор  <порядок байт>  <размер адресного пр-ва> <кол-во шагов>  <стартовый pc>  <образ> 

дальше начальный

closed 2015-12-14 by Jettatura