Zhikharev/selen #17: SIM, summary

Zhikharev, 2015-12-15:
Было бы круто если б в SUMMARY кроме количества исполненных инструкций так же выводилось количество инструкций обращение  память (возможно нужно будет для статистики). 

Можно даже пойти дальше и выводить статистику по всем инструкциям.

Будет ли у нас такой функционал?

Jettatura, 2015-12-15:
запросто

Jettatura, 2015-12-15:
а в каком виде:
```
SUMMARY:
шаг сделано:  200

таблица обращений к  памяти:
тип(чтение/запись)  инструкция(адрес/мнемоника)  адрес обращения   значение
    ...                                    ....                                                  ....                                ....
```

Zhikharev, 2015-12-15:
Я думаю просто некая статистика 
<code>
код инструкции(мнемоника)     количество 
</code>

closed 2016-03-09 by Jettatura