Zhikharev/selen #18: Remark on coding style

pbing, 2016-02-11:
1. always_ff, always_latch, always_comb... This allows simulators and synthesis tools better syntax checking.
2. Interfaces, at least for the WB bus ([for example](https://github.com/pbing/J1_WB/blob/master/rtl/if_wb.sv)). This will make top level wiring much easier. Also, assertions/checkers could be integrated into the interface.

Zhikharev, 2016-02-12:
Thank you for you reply. 
1) Sure, we should use system verilog features, but I am not sure that all simulators support them (fpga soft?). 
2) This is great for modeling and finding bugs, but again, what is about synthesis tools?

pbing, 2016-02-12:
1. My [J1 project](https://github.com/pbing/J1/blob/master/README.md) was developed with Altera Tools. Because the board uses an outdated FPGA the last revision was Quartus II 13.0SP1 was the last version that can be used. This version could not handle packed unions. I don't know about the latest version 15.1. The Modelsim simulator had no problems with SystemVerilog.

closed 2016-02-27 by Zhikharev