Zhikharev/selen #19: core_if_s

Zhikharev, 2016-02-16:
1) Добавить файл в core_new, файл должен соответствовать требованиям описания: выравнивание, шапка, гварды. 

2) Интерфейс.
input  clk  - нет вопросов
input  rst_n - нет вопросов

input	 if_enb - в каких случаях будет равен 0?
input	 if_kill - зачем данный сигнал? что должно происходить когда = 1 ?

input [31:0] if_jmp_brn_addr
input  if_jmp_brn 

input  if_pc_stop - зачем, если у нас есть enb?

output 	if_val
output [31:0]	if_pc
output [31:0]    if_pc_4

AlexBolotnikov, 2016-02-16:
Здравствуйте.

input if_enb,// = 0 когда столл (запрещяет изменение регистра)
input if_kill,// очищает регистр (когда =1 регистр сбрасывается) (вход rst_n фо фсех стадия я писал по умаолчанию его там быть не должно наверно он нужен для hazard control  когда в него поступает ресет то он ставит kill на все регистры а в pc ресет входит также системный)
//from haz 
input if_mux1_trn_pc_4_s,// select signsl for mux chousing next pc source
input if_pc_stop, впринцепи верно что есть enb но как я мыслю то enb для регистров между фазами а ps_stop он отдельно в регистр pc идет 
но по факту да всегда когда есть хотябы один enb = 0  то ps_stop = 1 это написанно в  hazard 
input[31:0] 
output if_val
output reg [31:0] if_lid_pc_out_reg, 
// to decode
output reg [31:0] if_pc,
output reg [31:0] if_pc_4 вопросы 

1)мне что делать с val для инстукций про который вы говорили 
2)какой у меня план работы что мне делать 
3)что с ошибками они есть про которые вы писали или нет (я еще отписался что я их уже исправлял(у меня build_rtl проходило нормально))
4)как мне поступать с always блоками делсть их как в system verilog или оставить 

> 1) Добавить файл в core_new, файл должен соответствовать требованиям описания: выравнивание, шапка, гварды.
> 2) Интерфейс.
> input  clk  - нет вопросов
> input  rst_n - нет вопросов
> input    if_enb - в [...]

AlexBolotnikov, 2016-02-19:
1)if_enb равен нулю в случаи stall pipeline 
2)if_kill очищает регистр if/dec от неправильной информации
3) да я согласен насчет if_pc_stop

Zhikharev, 2016-02-19:
1) добавь к регистрам суффикс _r
2) для jump используй наименования, что я предлагал выше: чтобы любой мог посмотреть и понять почему переключается счётчик

AlexBolotnikov, 2016-02-21:
1) тоеть заменить reg на r или я где то не дописал reg

Zhikharev, 2016-02-21:
Да заменить _reg  на _r, и, например, есть регистр pc, давай сделаем его pc_r.
Кроме того, так как мы будем синтезировать для Sparta-6, то придётся использовать ISE (Vivado не поддерживает spartan-6). А ISE не поддерживает system-verilog, поэтому сразу поменяй расширение файла с .sv на .v

AlexBolotnikov, 2016-02-21:
ок

AlexBolotnikov, 2016-02-21:
1)зачем делать лишную работe? какие основания заменять reg на r одно дело это сделать в if другое напрмер в decode.

closed 2016-02-27 by Zhikharev