Zhikharev/selen #20: SIM DPI bugs

Zhikharev, 2016-02-27:
Проверь коды операций (лучше сравнивать с riscv isa spec), сейчас я наблюдаю следующие расхождения между тестовым окружением и моделью:

```
UVM_INFO [     sl_core_scrb.sv(59) ] { 600.0ns } [b_env.core_scrb] [SCRB] SLTIU rs1=30 rs2= 2 rd=24 imm=000000e2
UVM_INFO [  sl_core_monitor.sv(61) ] { 600.0ns } [r_agent.monitor] [MON INSTR] item: { cop: RD  size: 'h4  addr: 'h210  data: 'he2f3c13  } 

[ISA SIM ERROR] I_R-type invalid func3 field:  opcode= 0x13 rd= 0x18 func3= 0x3 rs= 0x1e imm= 0xe2
```

Как декодирует тестовое окружение:
UVM_INFO [     sl_core_scrb.sv(59) ] { 600.0ns } [b_env.core_scrb] [SCRB] SLTIU rs1=30 rs2= 2 rd=24 imm=000000e2
Какие данные на шине (поле data):
UVM_INFO [  sl_core_monitor.sv(61) ] { 600.0ns } [r_agent.monitor] [MON INSTR] item: { cop: RD  size: 'h4  addr: 'h210  data: 'he2f3c13  } 

P.S. я правильно понимаю, что если step фэилится, то PC не переключается? Может лучше добавлять к нему 4, чтобы могли пройти далее?

Jettatura, 2016-02-27:
Это все у меня устарело и надо переделать, сейчас много чего декодируется неправильно. У меня на подходе большой коммит с простыми и правильным вещами, которые я слизал со Spike. Он давно зреет, и я думаю на неделе все заблестит)

Jettatura, 2016-03-05:
пробуй

Zhikharev, 2016-03-05:
UVM_INFO [     sl_core_scrb.sv(61) ] { 1800.0ns } [b_env.core_scrb] [SCRB] SRLI rs1=20 rs2=25 rd= 4 imm=00000419
UVM_INFO [  sl_core_monitor.sv(61) ] { 1800.0ns } [r_agent.monitor] [MON INSTR] item: { cop: RD  size: 'h4  addr: 'h240  data: 'h419a5213  } 

[ISA SIM ERROR] invalid instruction: 0x419a5213, op: 0x13 rd: 0x4 rs1:0x14 rs2:0x19 rm:0x5
"/media/sf_selen.git/verif/core/run//../uvm/scoreboard/sl_core_scrb.sv", 93: $unit::\sl_core_scrb::write_commit .unnamed$$_2: started at 1840000ps failed at 1840000ps
        Offending 'core_model::step()'
UVM_ERROR [     sl_core_scrb.sv(94) ] { 1840.0ns } [b_env.core_scrb] [MODEL] step failed!

Jettatura, 2016-03-05:
Проверь у себя, операнды вроде  правильные

Zhikharev, 2016-03-05:
а что проверить, симулятор же ругается

Ещё если выставить  model_params.mem_resize = 1; то симулятор может зависнуть

Zhikharev, 2016-03-05:
И ещё, конфигурирую следующим образом:
```
UVM_INFO [      sl_core_env.sv(41) ] { 0.0ns } [tb_env         ] [MODEL] Creating model...
UVM_INFO [      sl_core_env.sv(47) ] { 0.0ns } [tb_env         ] [MODEL] Initial model configuration
UVM_INFO [      sl_core_env.sv(48) ] { 0.0ns } [tb_env         ] [MODEL] pc_start 200
UVM_INFO [      sl_core_env.sv(49) ] { 0.0ns } [tb_env         ] [MODEL] mem_size 4096
UVM_INFO [      sl_core_env.sv(50) ] { 0.0ns } [tb_env         ] [MODEL] mem_resize 0
```

Но получаю сообщения типа:
[ISA SIM ERROR] write address refers to position out of memory: 0xfffff97e, avaible memory size: 0x400

Почему такой маленький размер?

Zhikharev, 2016-03-05:
По-поводу сдвига, исправил одну неточность: это не SRLI а SRAI

```
UVM_INFO [     sl_core_scrb.sv(61) ] { 1100.0ns } [b_env.core_scrb] [SCRB] SRAI rs1=29 rs2=19 rd= 3 imm=00000413
UVM_INFO [  sl_core_monitor.sv(61) ] { 1100.0ns } [r_agent.monitor] [MON INSTR] item: { cop: RD  size: 'h4  addr: 'h224  data: 'h413ed193  } 

[ISA SIM ERROR] invalid instruction: 0x413ed193, op: 0x13 rd: 0x3 rs1:0x1d rs2:0x13 rm:0x5
"/media/sf_selen.git/verif/core/run//../uvm/scoreboard/sl_core_scrb.sv", 93: $unit::\sl_core_scrb::write_commit .unnamed$$_2: started at 1140000ps failed at 1140000ps
        Offending 'core_model::step()'
```

Jettatura, 2016-03-06:
Я поправлю

Jettatura, 2016-03-09:
поправил

Zhikharev, 2016-03-09:
mem_resize всё ещё приводит к зависанию, что не так? В тестах я хочу использовать всё адресное пространство, это плохая идея?

Zhikharev, 2016-03-09:
Попробовал выделить 32 ГБ, процесс затянулся на какое-то время, но в итоге всё получилось.
Как у тебя организована память? Может в сторону ассоциативного массива посмотреть и создавать ячейки при первом обращении?

Jettatura, 2016-03-09:
да, сейчас все сделано очень просто: это массив в полный размер, и он долго выделяется.

Можно выделять память при первом обращении, плюс минус мегабайт(или задаваемое значение) от адреса. 
Но я думаю лучше сделать больше контроля: 
функция для управления памятью, объявить регион памяти: declareMemRegion(addr, size, attr)
attr это аттрибуты: разрешено чтение, запись, все вместе ().

С такой функцией можно делать MMU.

Jettatura, 2016-03-09:
и все обращения вне этих регионов, или с недопустимыми операциями будут выдавать ошибку

Jettatura, 2016-03-09:
я думаю это можно закрыть, а проблемы с памятью это уже другое

closed 2016-03-09 by Jettatura