Zhikharev/selen #21: XBAR bugs

Zhikharev, 2016-03-03:
Запуск:
export GIT_HOME=путь к проекту
make -f $GIT_HOME/verif/system/Makefile TB_DEFINES="+define+WAVES" build sim
создастся файл vcdplus, его можно в dve открыть. Или запуск сразу с gui:
make -f $GIT_HOME/verif/system/Makefile TB_DEFINES="+define+WAVES" build sim SIM_OPTS="-gui"

На коммутатор приходят запросы, он их перенаправляет в два слейв порта, хотя должен в один. Может с окнами что-то не так?

Ссылка на файл с диаграммами:
https://drive.google.com/file/d/0B3iBsagv95X1Sk05MEQ2VGNFa0k/view?usp=sharing

PetrakovP, 2016-03-03:
Не правильно настроены маски для адресов.
Если надо чтобы в 0ой слейв пападали обращения с 0x0h по 0x300h то дефайны должны быть такие :
`define WB_COM_S0_ABASE 32'h00000000
`define WB_COM_S0_AMASK 10'h3FF 
правда при этом в 0ой слейв будут так же пролезать адреса от 0x300 до 0x3FC. 
Те чтоб адреса для слейвов не пересекались у 1ого слейва ABASE должен начитнаться с 0x400

Возможно это не совсем правильное решение с дефайнами с моей стороны.
Если это не удобно можно придумать другие дафайны, вроде первого и последнего адресов пространства. А комутаторе будет проверка на то что адрес попадает в диапазон..

Zhikharev, 2016-03-03:
Понял. Я пока почти наобум дефайны задал, просто чтобы посмотреть, что хотя бы что-то работает

Zhikharev, 2016-03-04:
Я вроде обновил, но всё равно тоже самое. Посмотри пожалуйста doc/selen/selen_memory_map.docx и настрой два окна:одно под ram другое под io+rom

Zhikharev, 2016-03-04:
Хотя я уже обнаружил неточность:
parameter   S0_ADDR_BASE        =   `WB_COM_S0_ABASE;
parameter   S0_ADDR_MASK        =   `WB_COM_S0_AMASK;
parameter   S1_ADDR_BASE        =   `WB_COM_S0_ABASE;
parameter   S1_ADDR_MASK        =   `WB_COM_S0_AMASK;

используются дефайны с S0

closed 2016-03-05 by Zhikharev