Zhikharev/selen #22: Компиляция CORE

Zhikharev, 2016-03-03:
Обновил скрипты запуска.

Запуск с questasim:
export GIT_HOME=путь к проекту
export MTI_HOME=путь к папке с questasim
make -f $GIT_HOME/verif/core/run/questaMakefile MTI_HOME=$(MTI_HOME) build sim

Выдаются ошибки при компиляции.

Zhikharev, 2016-03-04:
Поправил скрипты. Сейчас запускается стадия элаборации: на ней много ошибок и предупреждений.

AlexBolotnikov, 2016-03-08:
bl@ubuntu:/home/selen/verif/core/run$ make -f questaMakefile build sim 
vlib work; \
	vmap work work; \
	vlog -l build.log -mfcu +define+UVM_NO_DPI -sv +incdir+.//../../tools/uvm-1.2/src ../../../rtl/core//opcodes.vh ../../../rtl/core//core_defines.vh ../../../rtl/core//core_reg_file.v ../../../rtl/core//core_dec_s.v ../../../rtl/core//core_alu.v ../../../rtl/core//core_exe_s.v ../../../rtl/core//core_if_s.v ../../../rtl/core//core_mem_s.v ../../../rtl/core//core_wb_s.v ../../../rtl/core//core_hazard_ctrl.v ../../../rtl/core//core_pipeline.v ../../../rtl/core//core_csr.v ../../../rtl/core//core_top.v +define+COMMIT_VAL_PATH=pipeline.pl_val_inst +define+REG_FILE_PATH=pipeline.core_dec_s.reg_file.reg_file +incdir+.//.. +incdir+.//../../common .//../testbench/core_model_dpi.sv .//../testbench/core_commit_if.sv .//../uvm/uvm.inc.sv .//../testbench/core_assembled.sv .//../testbench/core_tb_top.sv
** Warning: (vlib-34) Library already exists at "work".
Modifying modelsim.ini
QuestaSim vlog 6.6c Compiler 2010.08 Aug 23 2010
-- Compiling module core_reg_file
-- Compiling module core_dec_s
-- Compiling module core_alu
-- Compiling module core_exe_s
-- Compiling module core_if_s
-- Compiling module core_mem_s
-- Compiling module core_wb_s
-- Compiling module core_hazard_ctrl
-- Compiling module core_pipeline
-- Compiling module core_csr
-- Compiling module core_top
-- Compiling package core_model
-- Compiling interface core_commit_if
-- Compiling package uvm_pkg
** Warning: .//../../tools/uvm-1.2/src/base/uvm_event.svh(40): (vlog-2181) Use of a parameterized class uvm_event_callback as a type creates a default specialization.
-- Compiling interface core_if
** Warning: .//../uvm/items/rv32_typedefs.sv(9): (vlog-2263) Redefinition of macro: JALR.
-- Compiling package core_reg_file_v_unit
-- Importing package uvm_pkg
** Warning: .//../../tools/uvm-1.2/src/base/uvm_event.svh(40): (vlog-2181) Use of a parameterized class uvm_event_callback as a type creates a default specialization.
-- Importing package core_model
** Warning: .//../uvm/env/sl_core_env.sv(51): (vlog-2240) Treating stand-alone use of function 'init' as an implicit VOID cast.
** Warning: .//../../common/core/sl_core_slave_driver.sv(41): (vlog-2240) Treating stand-alone use of function 'randomize' as an implicit VOID cast.
-- Compiling interface core_if
-- Compiling module core_assembled
-- Compiling module core_tb_top

Top level modules:
	core_tb_top
vsim -novopt -sv_lib dpi_lib/uvm_dpi -sv_lib model_lib/model \
	-sv_seed 1 -do 'run -all' -l run.log  core_tb_top
Reading /opt/questasim_v6.6c/questasim/tcl/vsim/pref.tcl
 
все запускается и в конце в гуи опять пишет error loading design

Zhikharev, 2016-03-08:
А что пишет в gui до окончания элаборации? Какие ошибки?

AlexBolotnikov, 2016-03-08:
# //  QuestaSim 6.6c Aug 23 2010 Linux 3.16.0-30-generic
# //
# //  Copyright 1991-2010 Mentor Graphics Corporation
# //              All Rights Reserved.
# //
# //  THIS WORK CONTAINS TRADE SECRET AND 
# //  PROPRIETARY INFORMATION WHICH IS THE PROPERTY
# //  OF MENTOR GRAPHICS CORPORATION OR ITS LICENSORS
# //  AND IS SUBJECT TO LICENSE TERMS.
# //
# vsim -do {run -all} -l run.log -sv_seed 1 -novopt -sv_lib dpi_lib/uvm_dpi -sv_lib model_lib/model core_tb_top 
# (vsim-3763) SystemVerilog DPI cannot access file './dpi_lib/uvm_dpi.so'
# No such file or directory. (errno = ENOENT)
# Use the -help option for complete vsim usage.
# Error loading design

Zhikharev, 2016-03-08:
Очисть папку в которой собираешь дизайн, запустит с параметром SIM_OPTS="-c" и скопируй сюда вест лог от ввода команды, до завершения компиляции.

AlexBolotnikov, 2016-03-12:
Все проходит как я поимаю я залили лог в репозиторий 
[log.txt](https://github.com/Zhikharev/selen/files/170603/log.txt)

Zhikharev, 2016-03-13:
А файлы то обновлённые где?

closed 2016-03-13 by Zhikharev