Zhikharev/selen #24: [isa-sim] Реализовать менеджер памяти

Jettatura, 2016-03-09:


Zhikharev, 2016-03-25:
А что с этим?
И думаю тебе нужно трасировать обращения к тебе в файл, чтобы проще было отлаживать модель. А то сейчас что-то не работает, я скажу тебе что были то такие обращения, но разве ты мне поверишь? А так я бы тебе файл твой трассы скинул и всё.

Jettatura, 2016-03-26:
Это готово. Трассы будут в следующем комите

Zhikharev, 2016-03-27:
Т.е. как сейчас с этим работать? Я выставляю resize и всё автоматически будет работать или даже resize не нужен?

Jettatura, 2016-03-28:
интерфейс не поменялся, флаг resize только позволяет менять размер адресного пространства во время выполнения. 
что поменялось:  память выделяется по четыре килобайта при первой попытке записи не нуля по месту, те ничего не тормозит в начале, потому что ничего не выделяется, ну и памяти меньше весь симулятор требует.
По этому можно сначала установить размер памяти 4ГБ, флаг resize не ставить.
Или, если заранее не известно сколько  надо памяти, можно использовать флаг resize.

Надо удалить этот флаг, как считаешь, он и не нужен сейчас?

closed 2016-03-28 by Jettatura

Zhikharev, 2016-03-28:
Хм. Если заранее неизвестно, сколько памяти нужно, то как поступать, ещё раз?

Jettatura, 2016-03-28:
Если заранее неизвестно, сколько памяти нужно, то надо использовать allow resize.