Zhikharev/selen #25: register mapping

Jettatura, 2016-03-09:
riscv-gcc так размечает регистры:

````
const char* xpr_name[] = {
"zero", "ra", "sp", "gp", "tp", "t0", "t1", "t2",
"s0", "s1", "a0", "a1", "a2", "a3", "a4", "a5",
"a6", "a7", "s2", "s3", "s4", "s5", "s6", "s7",
"s8", "s9", "s10", "s11", "t3", "t4", "t5", "t6"
};

const char* fpr_name[] = {
"ft0", "ft1", "ft2", "ft3", "ft4", "ft5", "ft6", "ft7",
"fs0", "fs1", "fa0", "fa1", "fa2", "fa3", "fa4", "fa5",
"fa6", "fa7", "fs2", "fs3", "fs4", "fs5", "fs6", "fs7",
"fs8", "fs9", "fs10", "fs11", "ft8", "ft9", "ft10", "ft11"
};
````

у нас так?

Jettatura, 2016-03-09:
это не очень важно конечно, просто захотелось сделать это у себя в дизассемблере, а то у меня там до сих пор старая версия обозначений.

Zhikharev, 2016-03-09:
Думаю, в модели нужно соотв. стандартам, т.е. так

closed 2016-03-26 by Jettatura