Zhikharev/selen #26: [CORE] Запуск конвейера

Zhikharev, 2016-03-13:
Запуск: 
make -f $GIT_HOME/verif/core/run/Makefile UVM_HOME=/opt/uvm-1.2/ build  draft_test SIM_OPTS="+num_pkts=10 -gui"

От ядра идут запросы за инструкциями, но они не доходят до стадии commit, поэтому по окончанию моделирования ошибка:
UVM_ERROR [     sl_core_scrb.sv(129) ] { 1590.0ns } [b_env.core_scrb] [SCRB] pc_q is not empty (10)

AlexBolotnikov, 2016-03-17:
я все поправил и посмотрел на мой взгляд все нормально.

Zhikharev, 2016-03-19:
Во-первых, начальный адрес должен быть 0x200 согласно привилегированной документации

Zhikharev, 2016-03-19:
Во-вторых, изменение адреса не соответствует документации. Для обеспечения конвейерности два адреса должны быть всегда в полёте, смотри внимательно рисунок 5.

AlexBolotnikov, 2016-03-20:
Я поправил только не пойму правильно или нет все работает

Zhikharev, 2016-03-21:
Что значит не пойму? На диаграммах должен увидеть. 
Я посмотрел, неправильно.

AlexBolotnikov, 2016-03-21:
Я поправил на мой взгляд все правильно

Zhikharev, 2016-03-22:
Ты моделирование вообще запускал? 
Смотри, когда я предлагал сделать нормально ты отказался меня слушать. Сейчас почти две недели пытаешься поправить rtl так, чтобы запросы правильно выходили из ядра, такими темпами ничего хорошего не выйдет.

1) Адрес по сбросу не соотв. документации (уж это то можно было бы исправить)
2) Иксы на адресе

AlexBolotnikov, 2016-03-22:
[image]
[image]

1)Начальный адрес 0x200 как вы сказали я это поправил давно (`define PC_START  32'h200)
2)я не понимаю что не так с времеными диаграмами 
все как в документе сначала запрос потом через такт стол, потом стол реагируе на ack я не нахожу противоречий между моей временой диаграмой и диаграмой в reference manual
3) иксы только на шине адреса в кешь данных а не в кешь инструкция, что не так, я не понимаю, потому что инстукция не обращяестя в памать. Валидность запроса в кашь данных равна 0.

Zhikharev, 2016-03-22:
Ты уверен что ты закомитил свои изменения? 
Например, можешь посмотреть свои дефайны:
https://github.com/Zhikharev/selen/blob/master/rtl/core/core_defines.vh

Zhikharev, 2016-03-22:
1) COMMIT_VAL_PATH=pipeline.pl_val_inst  не должен быть в иксах, либо 0, либо 1
2) моделирование останавливается, потому что этот провод выходит на 1, и в ней остаётся, хотя на процессор была передана только 1 инструкция

AlexBolotnikov, 2016-03-22:
я починил ошибку.

AlexBolotnikov, 2016-03-22:
Notice: timing checks disabled with +notimingcheck at compile-time
Chronologic VCS simulator copyright 1991-2015
Contains Synopsys proprietary information.
Compiler version K-2015.09-SP1-1_Full64; Runtime version K-2015.09-SP1-1_Full64;  Mar 22 15:13 2016
VCD+ Writer K-2015.09-SP1-1_Full64 Copyright (c) 1991-2015 by Synopsys Inc.
The design has assertions or cover properties.
The assertion browser can be used to view them. Click on the assertion toolbar button or use the menu 'Window->Panes->Assertion' to open it.
The file '/home/synopsys/inter.vpd' was opened successfully.
UVM_INFO /media/sf_selen/verif/core/run//../testbench/core_tb_top.sv(46) @ 0.0ns: reporter [DBG] Set all core registers to zero...
UVM_INFO /media/sf_selen/verif/core/run//../testbench/core_tb_top.sv(68) @ 0.0ns: reporter [DBG] SEED = 1
UVM_INFO @ 0.0ns: reporter [RNTST] Running test draft_test...
UVM_INFO [      sl_core_env.sv(41) ] { 0.0ns } [tb_env         ] [MODEL] Creating model...
UVM_INFO [      sl_core_env.sv(48) ] { 0.0ns } [tb_env         ] [MODEL] Initial model configuration
UVM_INFO [      sl_core_env.sv(49) ] { 0.0ns } [tb_env         ] [MODEL] pc_start 200
UVM_INFO [      sl_core_env.sv(50) ] { 0.0ns } [tb_env         ] [MODEL] mem_size 42949
UVM_INFO [      sl_core_env.sv(51) ] { 0.0ns } [tb_env         ] [MODEL] mem_resize 0
UVM_INFO [  sl_core_seq_lib.sv(36) ] { 0.0ns } [lave_random_seq] [uvm_test_top.tb_env.core_data_agent.sequencer.sl_core_slave_random_seq] Start of sl_core_slave_random_seq
UVM_INFO [ sl_draft_seq_lib.sv(50) ] { 0.0ns } [r@@core_alu_seq] [uvm_test_top.tb_env.virtual_seqr.core_alu_seq] Start of core_alu_seq
UVM_INFO [ sl_draft_seq_lib.sv(62) ] { 190.0ns } [r@@core_alu_seq] [uvm_test_top.tb_env.virtual_seqr.core_alu_seq] End of core_alu_seq
UVM_INFO [     sl_core_scrb.sv(61) ] { 200.0ns } [b_env.core_scrb] [SCRB] SLL rs1=14 rs2=29 rd=24 imm=00000000
UVM_INFO [  sl_core_monitor.sv(61) ] { 200.0ns } [r_agent.monitor] [MON INSTR] item: { cop: RD  size: 'h4  addr: 'h200  data: 'h1d71c33  } 

                                      :X-
                                    :X###
                                  ;@####@
                                ;x######X
       TEST PASSED            -@#########$
                            .$###########@
                            =M############-
                           +##############$
                         .H############$=.
         ./:            .N##########M:.
      -+@NNN;          -##########M;
    -*M######         :#########M/
  -$M###########     :#########/
   ,:x###########:  =########$.
        ;H#########+#######N=
            ,+##############+
               /M#########@-
                 ;M######*
                   +###:
UVM_INFO [vm_report_server.svh(847) ] { 690.0ns } [reporter       ] [UVM/REPORT/SERVER] 
--- UVM Report Summary ---

** Report counts by severity
UVM_INFO :   13
UVM_WARNING :    0
UVM_ERROR :    0
UVM_FATAL :    0
** Report counts by id
[DBG]     2
[MODEL]     5
[MON INSTR]     1
[RNTST]     1
[SCRB]     1
[uvm_test_top.tb_env.core_data_agent.sequencer.sl_core_slave_random_seq]     1
[uvm_test_top.tb_env.virtual_seqr.core_alu_seq]     2

$finish called from file "/opt/uvm-1.2/src/base/uvm_root.svh", line 517.
$finish at simulation time 690.0ns
Simulation complete, time is 690000 ps.
Warning: [DVAP008] 
The source file 'sl_core_env.sv' cannot be found.

Zhikharev, 2016-03-23:
Почему ты запускаешь не то, что я в баге написал? Для запуска из темы, следующие ошибки:
Ошибки:
1) UVM_ERROR [     sl_core_scrb.sv(111) ] { 440.0ns } [b_env.core_scrb] [SCRB] PC compare failed. Received: 0000020c Expected: 00000208

адрес 208 проскакивает и не держится до ack

2) Инструкция
UVM_INFO [     sl_core_scrb.sv(61) ] { 400.0ns } [b_env.core_scrb] [SCRB] ANDI rs1=25 rs2=27 rd=21 imm=00000f1b

Потом на коммит стадии не происходит должного сравнения регистров

UVM_ERROR [     sl_core_scrb.sv(120) ] { 440.0ns } [b_env.core_scrb] [SCRB] REG[21] compare failed. Received: 00000000 Expected: fffff8ce

closed 2016-05-24 by Zhikharev