Zhikharev/selen #27: [XBAR] Deadlock

Zhikharev, 2016-03-19:
Ссылка на файл с диаграммами
Смотреть на модуль wb_perif_xbar.
На m0 порт приходят запросы за кэш строкой, коммуттатор перенаправляет их в s0. Затем приходит одиночный запрос, который должен быть перенаправлен на s1, но ничего не происходит.

Коммутатор правда по m0 выдаёт error, но я не понимаю почему. И не должен ли он вместе с error выдавать ack?

Zhikharev, 2016-03-19:
https://drive.google.com/file/d/0B3iBsagv95X1VE4xR3lXMGFxRHM/view?usp=sharing

Zhikharev, 2016-03-19:
По поводу трёх сигналов, уже нашёл ответ:
RULE 3.45
If a SLAVE supports the [ERR_O] or [RTY_O] signals, then the SLAVE MUST NOT assert
more than one of the following signals at any time: [ACK_O], [ERR_O] or [RTY_O].

Zhikharev, 2016-03-19:
Если что, окна настроены так:
	defparam wb_perif_xbar.S0_ADDR_BASE = 32'h0000_0000;
	defparam wb_perif_xbar.S0_ADDR_MASK = 16'h0fff;
	defparam wb_perif_xbar.S1_ADDR_BASE = 32'h0000_2000;
	defparam wb_perif_xbar.S1_ADDR_MASK = 16'h2fff;

Приходит запрос с адресом 32'h0000_2000

PetrakovP, 2016-03-22:
Попробуй маску для s1 сделать 16'h1fff. по-моему должно помочь. 
+ поправил небольшие недочеты, которые по вейвам увидел

Zhikharev, 2016-03-23:
Нет, так тоже не работает, вот запуск:
export GIT_HOME=<путь к репозиторию>
cp $GIT_HOME/verif/system/rom_image.v rom_image.v
make -f $GIT_HOME/verif/system/Makefile build sim SIM_OPTS="-gui" TB_DEFINES="+define+WAVES"

В сборке два коммутатора, ошибка на том, котоырй perif.
Диапазоны задаются в файле rtl/top/sele_top.sv

PetrakovP, 2016-03-23:
Точно для S1 поправил маску?

А то в комитах вижу:
	defparam wb_perif_xbar.S0_ADDR_BASE = 32'h0000_0000;
	//defparam wb_perif_xbar.S0_ADDR_MASK = 16'h0fff;
	defparam wb_perif_xbar.S0_ADDR_MASK = 16'h1fff;
	defparam wb_perif_xbar.S1_ADDR_BASE = 32'h0000_2000;
	defparam wb_perif_xbar.S1_ADDR_MASK = 16'h2fff;

Zhikharev, 2016-03-23:
Получается не то поправил.
Попробую отпишусь, значит должно быть

defparam wb_perif_xbar.S1_ADDR_BASE = 32'h0000_2000;
defparam wb_perif_xbar.S1_ADDR_MASK = 16'h1fff;

И это будет соответствовать диапазону: от 2000 до 2fff?

Zhikharev, 2016-03-23:
Вообще я сделал, всё заработало.

closed 2016-03-23 by Zhikharev