Zhikharev/selen #29: Программа для прототипа

Zhikharev, 2016-03-23:
Для запуска прототипа нужна программа, которая будет работать с gpio и моргать лампочкой.
Документация на gpio тут https://github.com/Zhikharev/selen/tree/master/doc/others

Zhikharev, 2016-03-28:
Алгоритм может быть следующий:
1) настроить 0 пин на вход, настроить параметр события (это будет внешняя кнопка)
2) дождаться события через полинг 
3) настроить 1 пин на выход (это будет лампочка)
4) используя таймер ядра инвертировать значение для пина 1 (мигание)

P. S. Можно на коленке написать, мы бы сразу провели моделирование

Jettatura, 2016-04-12:
firmware/sandbox/blink.c мои изыскания, предлагай детали

Zhikharev, 2016-04-13:
Таймер нужно посмотреть привилигированную архитектуру riscv, должны быть специальные системные инструкции.
Кстати умеешь пользоватьс inline ассемблером?

Jettatura, 2016-04-13:
Понял, умею.

Jettatura, 2016-04-13:
посмотри пожалуйста, как теперь?
ассемблерный листинг test.s вразумительный получается.
Стартап код?
можно наш простой 
```
la sp, stack_top
 j main
```
или надо что-нибудь поинтереснее?

Jettatura, 2016-04-14:
добавил скрипт линкера и startup код. Собирается вроде.

closed 2016-05-24 by Zhikharev