Zhikharev/selen #30: [isa-sim] Инструкции обращения в память

Zhikharev, 2016-03-25:
UVM_INFO [     sl_core_scrb.sv(61) ] { 120.0ns } [b_env.core_scrb] [SCRB] LW rs1= 0 rs2= 8 rd= 7 imm=00000148

UVM_INFO [     sl_core_scrb.sv(61) ] { 140.0ns } [b_env.core_scrb] [SCRB] LW rs1= 0 rs2=18 rd= 6 imm=000003f2

В память модели кладу данные:
UVM_INFO [     sl_core_scrb.sv(74) ] { 160.0ns } [b_env.core_scrb] [SCRB] Memmory acess cop=RDNC addr=148 data=6f7d4b4b

UVM_INFO [     sl_core_scrb.sv(61) ] { 160.0ns } [b_env.core_scrb] [SCRB] SW rs1= 0 rs2=25 rd=14 imm=00000b80

Ошибка:
UVM_ERROR [     sl_core_scrb.sv(120) ] { 180.0ns } [b_env.core_scrb] [SCRB] REG[7] compare failed. Received: 6f7d4b4b Expected: 00000000
UVM_ERROR [     sl_core_scrb.sv(92) ] { 180.0ns } [b_env.core_scrb] [SCRB] State compare failed!

Либо модель не исполняет инструкцию LW, либо данные не те берёт..

P.S. нужна реально твоя внутренняя трасса, но в отдельном файле.
P.P.S. потом можно будет добавить функцию получения времени моделирования, реализованную в SV

Zhikharev, 2016-03-29:
После послених правок ситуация седующая:
1) Идёт запрос за первой инструкцией
UVM_INFO [  sl_core_monitor.sv(61) ] { 120.0ns } [r_agent.monitor] [MON INSTR] item: { cop: RD  size: 'h4  addr: 'h200  data: 'h14802383  } 
Я эти данную кладу тебе в память: core_model::set_mem(item.addr, item.data) (почему ты это не стал трассировать?)
Затем при вызове step:
[ISA SIM ERROR] invalid instruction: 0x50431702, op: 0x2 rd: 14 rs1:6 rs2:4 rm:0x1

Трасса модели:
         0:	memory  READ; size 4 bytes; addr            0x200; value 0x50431702
         1:	memory  READ; size 4 bytes; addr            0x200; value 0x50431702
         2:	memory  READ; size 4 bytes; addr            0x200; value 0x50431702
         3:	memory  READ; size 4 bytes; addr            0x200; value 0x50431702
         4:	memory  READ; size 4 bytes; addr            0x200; value 0x50431702

Отличия от предыдущих запусков, я стал включать resize:
    model_params.pc_start   = 32'h0000_0200;
    model_params.mem_size   = 42949;
    model_params.verbose    = 1;
    model_params.mem_resize = 1;
    model_params.endiannes  = 1;

Jettatura, 2016-03-29:
можно трассу которая получится от сборки с моим последним коммитом, пожалуйста.

Zhikharev, 2016-03-29:
1) При запуске в консольном режиме есть проблемы в программном коде, по окончанию моделирования возникает ошибка
*** Error in `./simv': free(): invalid pointer: 0x000000000297b6d0 ***

Вырезка из stack trace с куском, указывающим на твой код
#12 0x000000000044a5e3 in ~page_t (this=<optimized out>, __in_chrg=<optimized out>) at /media/sf_selen.git//verif/core/run//../../../sim/memory.h:55

CoreTrace.txt пустой файл

Не знаю, с чем это связано. Может у тебя есть идеи?

2) При запуске с gui всё отрабатывает, прикладываю лог
[CoreTrace.txt](https://github.com/Zhikharev/selen/files/194121/CoreTrace.txt)

         0:	memory WRITE; size 4 bytes; addr            0x200; value 0x14802383
         1:	memory WRITE; size 4 bytes; addr            0x204; value 0x3f202303

        34:	memory  READ; size 4 bytes; addr            0x200; value 0x80a50902
        35:	fetch 0x80a50902 :   invalid

Jettatura, 2016-03-30:
Трасса совершенно фантастическая, записываем одно, позже читаем другое с того же адреса. Разберемся.

Jettatura, 2016-03-30:
Последний комит чинит все это, ошибка была совершенно дурацкая.

Теперь добавлен дамп памяти в конце, трасса для этого случая теперь выглядит так 
```
         0:	memory WRITE; size 4 bytes; addr 0x00000200; value 0x14802383
         1:	memory WRITE; size 4 bytes; addr 0x00000204; value 0x3f202303
         2:	memory  READ; size 4 bytes; addr 0x00000200; value 0x14802383
         3:			||-- 0x00000200	 0x14802383	    LW	 t2, [zero + 0x148]
         4:	register  READ; id 0 -> zero; value  0x00000000
         5:	memory  READ; size 4 bytes; addr 0x00000148; value 0x00000000
         6:	register WRITE; id 7 -> t2; value  0x00000000; diff 0 (0)
```
Вот теперь если что то сломается есть тест воспроизводилка sim/dpi/test.c.
Вот.

Zhikharev, 2016-03-31:
>> Так же model_params.endiannes = 1; не надо, это big endian, надо ставить 0, little endian

На что параметр влияет? Я запускал только алу инструкции - данный параметр не менял ничего.

Zhikharev, 2016-03-31:
Ты бы писал вызовы функций dpi и аргументы которые тебе передаются.
И такое ощущение, что когда нужно завершать моделирование твоя модель зависает, может деструктор какой нужно в интерфейс сделать?

Jettatura, 2016-03-31:
это только так получается что не влияет: пока ты читаешь и пишешь слова, выровненые по четыре байта все работает и так и так: записали слово, прочитали его же и не важно как оно в памяти лежит. А попробуй интсрукцию которая читает один байт с какого нибудь невыровненого адреса, куда до этого было записано слово: все изменится.

А куда мне писать вызовы, в трассу, в консоль?

Jettatura, 2016-03-31:
я думаю она зависает потому что происходит дамп памяти в файл. Я сделаю  опции отключения дампа и разыные флаги настройки трассы.

Jettatura, 2016-03-31:
или даже будет функция вызывающая дамп памяти в файл

Zhikharev, 2016-04-01:
Есть такая проблема:
         0:	memory WRITE; size 4 bytes; addr 0x00000200; value 0xa2202d83
         1:	memory WRITE; size 4 bytes; addr 0x00000204; value 0x55e02183
         2:	memory WRITE; size 4 bytes; addr 0xfffffa22; value 0x6f7d4b4b
         3:	memory WRITE; size 4 bytes; addr 0x00000208; value 0xfbd029a3
         4:	register  READ; id 0 -> zero; value  0x00000000
         5:	register  READ; id 1 -> ra; value  0x00000000
         6:	register  READ; id 2 -> sp; value  0x00000000
         7:	register  READ; id 3 -> gp; value  0x00000000
         8:	register  READ; id 4 -> tp; value  0x00000000
         9:	register  READ; id 5 -> t0; value  0x00000000
        10:	register  READ; id 6 -> t1; value  0x00000000
        11:	register  READ; id 7 -> t2; value  0x00000000
        12:	register  READ; id 8 -> s0; value  0x00000000
        13:	register  READ; id 9 -> s1; value  0x00000000
        14:	register  READ; id 10 -> a0; value  0x00000000
        15:	register  READ; id 11 -> a1; value  0x00000000
        16:	register  READ; id 12 -> a2; value  0x00000000
        17:	register  READ; id 13 -> a3; value  0x00000000
        18:	register  READ; id 14 -> a4; value  0x00000000
        19:	register  READ; id 15 -> a5; value  0x00000000
        20:	register  READ; id 16 -> a6; value  0x00000000
        21:	register  READ; id 17 -> a7; value  0x00000000
        22:	register  READ; id 18 -> s2; value  0x00000000
        23:	register  READ; id 19 -> s3; value  0x00000000
        24:	register  READ; id 20 -> s4; value  0x00000000
        25:	register  READ; id 21 -> s5; value  0x00000000
        26:	register  READ; id 22 -> s6; value  0x00000000
        27:	register  READ; id 23 -> s7; value  0x00000000
        28:	register  READ; id 24 -> s8; value  0x00000000
        29:	register  READ; id 25 -> s9; value  0x00000000
        30:	register  READ; id 26 -> s10; value  0x00000000
        31:	register  READ; id 27 -> s11; value  0x00000000
        32:	register  READ; id 28 -> t3; value  0x00000000
        33:	register  READ; id 29 -> t4; value  0x00000000
        34:	register  READ; id 30 -> t5; value  0x00000000
        35:	register  READ; id 31 -> t6; value  0x00000000
        36:	memory  READ; size 4 bytes; addr 0x00000200; value 0x00000000
        37:			||-- 0x00000200	 0x00000000	invalid	
        38:	memory WRITE; size 4 bytes; addr 0x0000055e; value 0x36e9840f
        39:	memory WRITE; size 4 bytes; addr 0x0000020c; value 0x42e04203

Jettatura, 2016-04-01:
повторю

Jettatura, 2016-04-01:
Спасибо, теперь этот случай выглядит так:
```
         0:	[DPI CALL]     set_mem addr 0x20000000, data 0xa2202d83
         1:	[memory]       WRITE; size 4 bytes; addr 0x20000000; value 0xa2202d83
         2:	[DPI CALL]     set_mem addr 0x20400000, data 0x55e02183
         3:	[memory]       WRITE; size 4 bytes; addr 0x20400000; value 0x55e02183
         4:	[DPI CALL]     set_mem addr 0xfffffa22, data 0x6f7d4b4b
         5:	[memory]       WRITE; size 4 bytes; addr 0xfffffa22; value 0x6f7d4b4b
         6:	[DPI CALL]     set_mem addr 0x20800000, data 0xfbd029a3
         7:	[memory]       WRITE; size 4 bytes; addr 0x20800000; value 0xfbd029a3
         8:	[DPI CALL]     step 
         9:	[memory]       READ ; size 4 bytes; addr 0x20000000; value 0xa2202d83
        10:	[fetch]        	-|-->	0x20000000	 0xa2202d83	    LW	s11, [zero + 0xfffffa22]
        11:	[register]     READ ; id 0 -> zero; value  0x00000000
        12:	[memory]       READ ; size 4 bytes; addr 0xfffffa22; value 0x6f7d4b4b
        13:	[register]     WRITE; id 27 -> s11; value  0x6f7d4b4b; diff 1870482251 (0x6f7d4b4b)
```
В результате последней правки симулятор сильно прибавил,  поменялась cборка: sim/isa/src.cpp теперь переехало в sim/model/src.cpp, инициалиализация тоже поменлась, смотри файл sim/dpi/core_model_dpi.h.
Так же по этим двум кейсам  зародились мои собственные тесты:  sim/dpi/test.c, поэтому буду рад другим кейсам, сообщениям о багах и другому фидбеку.
Это надо переоткрыть если что.

Jettatura, 2016-04-01:
verbosity_level пока не работает так как надо, пока уровень только один, но скоро все будет

Zhikharev, 2016-04-02:
Давай issue будет закрывать тот, кто открыл, после проверки :)

Zhikharev, 2016-04-02:
Трасса твоя перестала в файл сохраняться

Jettatura, 2016-04-02:
Надо ставить в параметрах toFile, посмотри

Zhikharev, 2016-04-02:
Увидел, что ты добавил параметры в dpi, интерфейс. Попробую

Zhikharev, 2016-04-02:
474:	[DPI CALL]     set_mem addr 0xfffff9b2, data 0x10c54dc2

       475:	[memory]       WRITE; size 4 bytes; addr 0xfffff9b2; value 0x10c54dc2

       543:	[DPI CALL]     step 
       544:	[memory]       READ ; size 4 bytes; addr 0x21800000; value 0x9b200703
       545:	[fetch]        	-|-->	0x21800000	 0x9b200703	    LB	 a4, [zero + 0xfffff9b2]
       546:	[register]     READ ; id 0 -> zero; value  0x00000000
       547:	[memory]       READ ; size 1 bytes; addr 0xfffff9b2; value 0xffffffffffffffc2
       548:	[register]     WRITE; id 14 -> a4; value  0xffffffc2; diff -62 (0xffffffc2)

UVM_ERROR [     sl_core_scrb.sv(123) ] { 1520.0ns } [b_env.core_scrb] [SCRB] REG[14] compare failed. Received: ffffffc5 Expected: ffffffc2
UVM_ERROR [     sl_core_scrb.sv(94) ] { 1520.0ns } [b_env.core_scrb] [SCRB] State compare failed!

Ситуация в следующем:
1) смещение в адресе 2, значит запрашивается второй байт
2) на шине данные 10 C5 4D C2, второй байт это C5
3) Модель должна учитывать адрес, когда идут байтовые обращения
4) Вместо ffffffc2, реально должно быть значение ffffffc5

Zhikharev, 2016-04-04:
Мне нужно будет set_mem делать для выровненного адреса по 4 байта

Zhikharev, 2016-04-05:
Тут всё работает.

closed 2016-04-05 by Zhikharev