Zhikharev/selen #33: [isa-sim] SLL

Zhikharev, 2016-04-06:
UVM_INFO [     sl_core_scrb.sv(61) ] { 17500.0ns } [b_env.core_scrb] [SCRB] SLL rs1=30 rs2=30 rd=16 imm=00000000

rs1 = 0xfffff85f
Сдвиг на 1f (5 младших бит), получается  80000000

UVM_ERROR [     sl_core_scrb.sv(124) ] { 17540.0ns } [b_env.core_scrb] [SCRB] REG[16] compare failed. Received: 80000000 Expected: 00000001

     12240:	[memory]       READ ; size 4 bytes; addr 0x000004a0; value 0x01ef1833
     12241:	[fetch]        	-|-->	0x000004a0	 0x01ef1833	   SLL	 a6,  t5, t5
     12242:	[register]     READ ; id 30 -> t5; value  0xfffff85f
     12243:	[register]     READ ; id 30 -> t5; value  0xfffff85f
     12244:	[register]     WRITE; id 16 -> a6; value  0x00000001; diff 1 (0x1)

Zhikharev, 2016-04-06:
У тебя походу для sll и srl перепутаны стороны сдвигов?

Jettatura, 2016-04-06:
да, это так

Jettatura, 2016-04-06:
починил

Zhikharev, 2016-04-07:
12239:	[DPI CALL]     step 
     12240:	[memory]       READ ; size 4 bytes; addr 0x000004a0; value 0x01ef1833
     12241:	[fetch]        	-|-->	0x000004a0	 0x01ef1833	   SLL	 a6,  t5, t5
     12242:	[register]     READ ; id 30 -> t5; value  0xfffff85f
     12243:	[register]     READ ; id 30 -> t5; value  0xfffff85f
     12244:	[register]     WRITE; id 16 -> a6; value  0x00000001; diff 1 (0x1)

Всё равно такое же результат

closed 2016-04-07 by Zhikharev