Zhikharev/selen #34: [XBAR] Окна

Zhikharev, 2016-04-08:
defparam wb_cpu_xbar.S0_ADDR_BASE = 32'h0000_0000;
	defparam wb_cpu_xbar.S0_ADDR_MASK = 16'h2fff;
	defparam wb_cpu_xbar.S1_ADDR_BASE = 32'h0001_0000;
	defparam wb_cpu_xbar.S1_ADDR_MASK = 32'h0004_ffff;

На запросы с адресами 
2 0200 
2 0204
..
2 0218

Приходит ответ m0_wb_err_i

Почему?

PetrakovP, 2016-04-11:
S1 BASE и MASK как-то несогласованы =\

Используются эти параметры только в одной строчке, дял определения пападания запроса в один их слейвов:
assign m_addr_s1_hit = (m_wb_addr_o & ~({WB_ADDR_WIDTH{1'b0}} | S1_ADDR_MASK)) == S1_ADDR_BASE;

Как это работает: 
1)от адреса с wb шины отсекается(зануляется) маска. Те все биты которые были равны 1'b1 в  маске зануляют биты адреса
 (m_wb_addr_o & ~({WB_ADDR_WIDTH{1'b0}} | S1_ADDR_MASK)) 
2) результат предыдушей операции сравнивается с базой
== S1_ADDR_BASE

Если адрес попал куда-то, то он уходит в нужный слейв. Если не попал - выдается err_i.

Вообще я пытался сделать это как настройка базовых адресов в конфиг пространстве pci, где MASK - это битовая шкала RO битов, а BASE это само значение базового адреса(уже с учетом RO битов).

Касательно нашей ситуации: 
после наложения маски на адрес получается - 2_0000, которое никак не равно 1_0000 в базе. Отсюда и err_i.

Честно говоря, я задумывал что значения MASK будут представлять собой непрерывную последовательность единичек. Поэтому мне непонятны значения типа 2fff или 4ffff

PetrakovP, 2016-04-11:
Несогласованые S1 MASK и BASE:
Изначально я думал, что через mask и base будут задаваться окна адресов подобно регистрам базовых адресов в pci конфиг пространстве. Где mask являлось бы непрерывной последовательностью единичек, показывающей количество ReadOnly бит в регистре, а base - былобы само значение базового регистра с учетом RO бит. Поэтому мне не совсем понятны значения типа 2fff или 4_ffff.

Вообще эти параметры используются только в одной строчке:
assign m_addr_s1_hit = (m_wb_addr_o & ~({WB_ADDR_WIDTH{1'b0}} | S1_ADDR_MASK)) == S1_ADDR_BASE;
Т.е. у адреса на wb шине зануляются MASK биты, и потом результат сравнивается с BASE.
Если значение адреса никуда не попадает, то это err_i.
В нашем случае после нуления MASK бит получается 2_0000, которое никак не равно 1_0000 в BASE. Отсюда и err_i

Zhikharev, 2016-04-12:
Понятно.
Получается я не могу сделать окно с адресами 

1_000 
4_ffff,

 но могу сделать окно
10_0000
1f_ffff

PetrakovP, 2016-04-12:
Выходит так.
Если это не удобно можно переписать. И например, указывать верхнюю и нижнюю границы окна. 
И сделать простую проверку, что адрес попадает в этот промежуток. Наверное так будет проще и понятнее, нет?

Zhikharev, 2016-04-13:
Скорее всего ты прав, что проще делать два адреса. Но пока не стоит ничего переделывать.
Я пытаюсь найти время, чтобы скомпилировать то, что есть для ПЛИС и попробовать получить каку-либо рабочую версию. 
После уже будем думать :)

closed 2016-04-13 by Zhikharev